rppicomidi/EZ_USB_MIDI_HOST #1: Do some configuration at run-time

rppicomidi, 2024-05-13:
Arduino libraries do not include files from the application sketch directory, and touching library configuration files in the libraries directory changes the configuration for all applications. Try instead to do at least some configuration at run-time.

rppicomidi, 2024-05-21:
In order to use the run-time changes to the usb_midi_host library, I had to change the API for this library. Version 2.0.0 made the class libraries C++ templates. You can now subclass the default settings structure to override the settings you are most likely to need to change. See the documentation for more details.

@VocasoGK Please test version 2.0.0 of the EZ_USB_MIDI_HOST library. As I understand it, your project is using this library. I believe I fixed [rppicomidi/usb_midihost#13](https://github.com/rppicomidi/usb_midi_host/issues/13)  in an Arduino friendly way. Please open a new issue in EZ_USB_MIDI_HOST if you have problems.

closed 2024-05-21 by rppicomidi

VocasoGK, 2024-05-22:
Sure do! Will update when I have the time!