rppicomidi/EZ_USB_MIDI_HOST #4: EZ_USB_MIDI_HOST_PIO_example.ino Does not Recognize MIDI Devices Attached to Host-Connector

SoerenSofke, 2024-06-28:
Hi,

I really appreciate this library, as it provides an excellent starting point for my own project, which is to create a MIDI message filter.  I intend to instrument [EZ_USB_MIDI_HOST](https://github.com/rppicomidi/EZ_USB_MIDI_HOST) on my [Adafruit Feather RP2040 with USB Type A Host](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host), connect a MIDI device to the USB Type A Host connector on the Feather and a PC with a software synthesizer on the USB Type C connector on the Feather.

However, I can not even make the given example, e.g. [EZ_USB_MIDI_HOST_PIO_example.ino](https://github.com/rppicomidi/EZ_USB_MIDI_HOST/blob/2.0.0/examples/arduino/EZ_USB_MIDI_HOST_PIO_example/EZ_USB_MIDI_HOST_PIO_example.ino) run as expected.

**My development environment**

* Fedora Linux, Version 40
* Arduino IDE, Version 2.3.2
* EZ_USB_MIDI_HOST, Version 2.0.0
* Adafruit Feather RP2040 with USB Type A Host

**Things that work as expected**

* Compilation and upload of [EZ_USB_MIDI_HOST_PIO_example.ino](https://github.com/rppicomidi/EZ_USB_MIDI_HOST/blob/2.0.0/examples/arduino/EZ_USB_MIDI_HOST_PIO_example/EZ_USB_MIDI_HOST_PIO_example.ino) without any modifications
* Blinking of red LED (LED_BUILTIN) on Feather, when Feather is connected to PC via USB Type C connector
* Serial monitor in Arduino IDE output diagnostic strings, e.g.
`19:34:36.589 -> EZ USB MIDI HOST PIO Example for Arduino`
`19:34:36.589 -> Core1 setup to run TinyUSB host with pio-usb`
* Green LED indicating 5V power supply on USB Type A Host connector

**Things that do NOT work as expected**

* Recognition of MIDI controller by [EZ_USB_MIDI_HOST](https://github.com/rppicomidi/EZ_USB_MIDI_HOST)  and provisioning of subsequent functionality, e.g. receiving and processing MIDI events. 

I would expect the serial monitor to output diagnostic strings on `onMIDIconnect` and `onMIDIdisconnect` events whenever I connect or disconnect a MIDI device using the Feathers USB Type A connector. However, this is not the case even though my MIDI devices are powered  and working. I have tested with two different MIDI devices, e.g. [Arturia MicroLab](https://www.arturia.com/products/hybrid-synths/microlab/overview) and [ESI Xjam](https://www.esi-audio.com/products/xjam/) without success.

**How to proceed**

* I would not rule out mistakes in compiling the sources or configuring the Feather.  So please make suggestions and challenge my approach.
* I would not exclude that my expectations are wrong. So please challenge them as well.
* If more information is needed to help me run the example, please give me detailed instructions to follow.  My project really depends on this library, and I would like to give you the best feedback I can.

Thanks, Sören

rppicomidi, 2024-07-01:
@SoerenSofke Thank you for taking the time to report this. Please tell me what version of the Pico-PIO-USB library you are using. If you are using 0.6.0, please downgrade to 0.5.3. There is a [known issue](https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/122) with 0.6.0

SoerenSofke, 2024-07-01:
Dear @rppicomidi 

Thanks for your quick reply. Indeed, version 0.6.0 of the Pico-PIO-USB library was active in my project, and I downgraded to version 0.5.3 following your suggestion. The situation improved in the sense that the MIDI controller connected to the USB Type A Host connector of the Feather board was now successfully recognized:

`21:27:11.549 -> EZ USB MIDI HOST PIO Example for Arduino`
`21:27:11.549 -> Core1 setup to run TinyUSB host with pio-usb`
`21:27:12.065 -> MIDI device at address 1 has 1 IN cables and 1 OUT cables`
`21:27:12.808 -> C1: Note on#51 v=16`
`21:27:13.001 -> C1: Note on#51 v=0`
`21:27:13.130 -> C1: Note on#51 v=28`
`21:27:13.259 -> C1: Note on#51 v=0`
`21:27:13.422 -> C1: Note on#51 v=11`
`21:27:13.583 -> C1: Note on#51 v=0`

On the other hand, there are some new findings that I would like to discuss in a new issue, if you agree.

Best, Sören

SoerenSofke, 2024-07-02:
**Solution**

EZ_USB_MIDI_HOST, version 2.0.0 with dependent library Pico-PIO-USB library in version 0.5.3 recignizes MIDI devices connected to [Adafruit Feather RP2040 with USB Type A Host](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host).

Downgrade of Pico-PIO-USB library required.

closed 2024-07-02 by SoerenSofke

rppicomidi, 2024-07-08:
@SoerenSofke For what it is worth, even version 0.6.0 of the Pico PIO USB library works for me with Adafruit TinyUSB 3.3.1  if the "Tools->Optimize" menu item is set to "Optimize Even More (-O3)" The bug report says only -Os is known to not work.