rppicomidi/EZ_USB_MIDI_HOST #6: Examples for connecting to USB hub?

DisasterAreaDesigns, 2025-01-06:
Hey there!  This library is great, thanks so much for your hard work on it.  Would it be possible to get a simple example for connecting multiple devices via a USB hub?  I'm on Arduino FWIW.

rppicomidi, 2025-01-09:
@DisasterAreaDesigns C++ and Arduino are pretty much the same. I have three C++ examples
https://github.com/rppicomidi/midi2usbhub
https://github.com/rppicomidi/midi2usbhub-web
https://github.com/rppicomidi/midi2piousbhub
Simple examples are a bit tricky because you have to deal with more than one device and define how to send MIDI data from one to the other, and you have to deal with the possibility that one or more of the devices you expect to send data to is disconnected. All of this is made more complex because each USB device can have up to 16 virtual MIDI IN ports and 16 virtual MIDI OUT ports.

Do you have a specific simple example in mind? If you do, please provide details here; I am happy to give you guidance on how to write it. If not, please have a look at the examples I have provided.

rppicomidi, 2025-01-10:
@DisasterAreaDesigns I just realized that none of the examples I cited actually use this library. They use the usb_midi_host library without this wrapper.

To use this library with multiple devices, you need to keep track of the device address of each device connected to the hub. You get that information in the mount callback, called onMIDIconnect in the example. That callback also returns how many virtual cables each device has. The code has to keep track of how to route data among the devices. The code needs to call getInterfaceFromDeviceAndCable() for each device address and each cable that you want to route.

As you can see, this gets complex pretty fast once you are dealing with more than one device. It is why I need a description of your application so I can help you develop the code you need.

DisasterAreaDesigns, 2025-01-10:
As an example, my thought was something like the basic sketch you have already posted.  Enumerate any connected devices, print a message for debug, and then send notes to each connected device as a test.  Any data coming in from the devices gets piped back to the host but not between devices.

rppicomidi, 2025-01-13:
@DisasterAreaDesigns I decided to task switch to this issue. I pushed version 2.1.0 of this library. There is an Arduino example that supports multiple devices and virtual MIDI IN cables.

Thank you for submitting this feature request. I helped me find a bug in plug/unplug management that I never would have found otherwise.

If you have problems or need help with this example, please open a new issue. Thanks again for taking the time to submit this issue.

fikryfadh, 2025-01-13:
@rppicomidi im curious instead using hub, is this library can support to add more physical usb host? 

[](https://github.com/sekigon-gonnoc/Pico-PIO-USB/discussions/72)
refer to this it can be configured by function `pio_usb_host_add_port` so it can initialize 2 port?

recently i tried with usb hub but the problem is device address will sort by first plugged device into hub port, wondering to add different host made device address can be static by each port.

rppicomidi, 2025-01-16:
@fikryfadh This library supports exactly one host port, so to answer your question, no, the library cannot currently support more than one physical USB port.

To address your issue, once it has the device address, your program  can call `getDevFromDevAddr()` to get a pointer to the `EZ_USB_MIDI_HOST_Device` object, and then use that pointer to call `get_vid()` and `get_pid()` so you know which device is which.

If you wish to discuss further, please open a new issue. Asking for a hub support example is what issue #6 is about. I did that and closed the issue. Adding a second host port is a new feature request, so it needs a new issue.

rppicomidi, 2025-01-30:
@DisasterAreaDesigns I finally came up with a work-around for the difficulty identifying devices. If you pull the latest code, you will note that the regular examples all support up to 4 devices connected to a hub. The examples show how the application can better use the library to track what devices are connected. Also, I added an API for getting the VID, PID, and device string descriptors for each connected device. If these changes help you, please let me know and close this issue.

DisasterAreaDesigns, 2025-02-01:
@rppicomidi thanks!  I've been pulling the VID / PID the hard way, this makes it a lot simpler.  Let me do some testing but I think this is 100% helpful, thank you.

rppicomidi, 2025-02-11:
@DisasterAreaDesigns Have you been able to do your testing? Is it OK to close this issue?

DisasterAreaDesigns, 2025-02-28:
Yes, I think I’m all good here.  Will let you know if I run into any issues.

disasterareadesigns.com

rppicomidi, 2025-03-04:
OK. I will close this issue.