rppicomidi/EZ_USB_MIDI_HOST #10: Can't Detect or connect to DIY usb device

rudiars, 2025-02-03:
hi, I have tried all versions of the library. is running according to its function. I tried connections with M-AUDIO, Behringer, Korg, and Yamaha. everything can work well for input and output. but if I try with USB Midi Devices made with microcontrollers such as ATmega16u2 and STM32F103C8T6 everything doesn't work. even though on the PC computer it is detected and working well. please provide a solution. Thank You

rppicomidi, 2025-02-03:
@rudiars Thank you for filing this issue and I am sorry this is not working for you. I am happy to try to help and will fix software issues in this library if we can find them. Please provide me with more information.

1. Are you using this library with Arduino or C++?
2. What program are you using to test the library? Is it an example code in this project, or your own?
3. Tell me more about your hardware. What board is running this code? Are you using the RP2040 native USB port for the host, or are you using the Pico-PIO-USB for the host? Do you connect your DIY board directly to the host, or through a hub? If a hub, is the hub powered or unpowered?
4. What troubleshooting steps have you tried already? Have you seen [this document](https://github.com/rppicomidi/pico_usb_host_troubleshooting)? Or [this document](https://github.com/rppicomidi/pico_usb_host_troubleshooting)? For me, issues like this turned out to be power supply or wiring related, but I am not ruling out a software issue in this driver.
5. If C++, please build configure the build with LOG=2 (`cmake -DLOG=2`) and rebuild the code. Then record the serial port log output to file, and and attach it to this issue.
6. I do not have the microcontrollers you listed in your issue, but I have the STM32F072 Discovery kit board. Will your STM32F103 example run on it? If so, please send to me a link to the software I can try.

rudiars, 2025-02-03:
I'm sorry sir. my bad. Last try it's work well. may be it's power issues.  
I'm using arduino IDE and from your example (non pio), USB HUB, and 4 USB device (Maudio oxygen,  Behringer BCF2000, STM32F103C8T6 usb midi device, and Atmega16u2 (moco lufa).
all work great now. thanks a lot for this library. regards

rppicomidi, 2025-02-03:
@rudiars So glad it is working for you. Nothing to be sorry about. Please do not hesitate to reach out if you encounter other issues. I am closing this now.

closed 2025-02-03 by rppicomidi